lab11/signpost #1: Backplane updates

bradjc, 2016-09-09:
So I don't forget:

- Numbers in silkscreen for modules
- Move part labels off of vias (korea is ok when the vias are small, these vias are big)
- Fix U30 package (and the other 5)
- Add jumper spot for !backplane_reset to VCC
- Minimize plane area (?)

ppannuto, 2016-09-09:
* USB data muxes have wrong control signals. Should be fed with `MODn_USB_!EN`, not `MODn_ISO_USB`

bradjc, 2016-09-09:
I also think we should revisit this somewhat bizarre decision to reset by default. Do we really want the controller to say "If I can't work then nothing can work!!! Muwahahahaha!"?

ppannuto, 2016-09-09:
It wasn't so malicious as that sounds. The idea behind it was protection of
the system. If the controller resets itself, dies, or w/e, then while the
controller is out of commission a wayward module can't bleed the system
dry.

What would you prefer the default behavior to be if there's no active
controller, turn everything on?

bradjc, 2016-09-09:
In the case that the controller is gonezo, then everything is broken and I don't think it matters what happens. But in case that controller resets briefly, everything would break, even in the case the controller comes back in a few milliseconds. Things like USB to linux, and I2C to radio can all keep happening while the controller resets. And the power supply should still be aggregating the energy of each module.

ppannuto, 2016-09-09:
That's almost already the situation.

Because all of the modules are on GPIO extenders, they wouldn't change state when the controller goes away. The missing bit is that `!BACKPLANE_RESET` will trip because the controller isn't holding it any more (which will then cause all the GPIO extenders to reset).

Perhaps the right solution then is to latch/buffer `!BACKPLANE_RESET` the same way. That way the signpost still "boots up" to a sane state when all power is lost, but if the controller disappears for a bit things will keep on keeping on (and the controller can read all the GPIO extenders to learn the state on reset).

bradjc, 2016-09-09:
What about just pulling !RESET high and calling it a day? Who knows what state everything is in during reset, but everything should be fine in the default, just resetted state. The controller can reset everything if it wants. All the modules will still default to disabled.

Also, it seems like the unused port should have pull downs: https://e2e.ti.com/support/interface/digital_interface/f/130/p/367935/1295211#1295211

ppannuto, 2016-09-11:
Currently, the debug LEDs for the backplane live in the corner on each module, but are covered by the backplane case when it's installed. 

The original thoughts for this were
  - You likely won't populate the debug LEDs on a deployed board b/c they burn power
  - There are no components anywhere under modules so module creators can consider the full vertical space under the module as theirs

Thinking more
  - It sucks a little that you don't have the option for debug LEDs and case
  - It's unlikely that an smd LED and an 0603 resistor are going to break the bank for vertical space for module makers (though it does become something they may have to think about in corner cases)

I'm leaning towards moving these more in the middle of the module space but wanted to see if anyone thought that was a terrible idea for some reason

bradjc, 2016-09-12:
At first I was opposed to this, then for it, and now I'm against it again. While I agree having the LEDs visible inside the case would be useful (particularly for a demo), they are only useful for populated modules, but once the module is populated you won't be able to see the LEDs anymore.

ppannuto, 2016-09-12:
Yeah, that's the same back and forth I went through the first time. Eventually concluded that having unpopulated parts visible wasn't a great aesthetic, happy to leave that conclusion.

ppannuto, 2016-09-12:
@adkinsjd can you send me any hole / mounting / plating requirements for the new power connectors you're looking at? I just need enough dimensions to put holes in the right place on the PCB

ppannuto, 2016-09-13:
@adkinsjd I removed the jumper terminals and put one hole at `(1.749, 5.751) drill 1.125`.

Do I need to do anything else?

ppannuto, 2016-09-20:
Most recent commit removes a lot of metal. Hopefully that should make things more manageable.

@bradjc is there anything worth trying to do to the footprint of U20-25 to make that part more solderable? Smaller planes will help, but it's necessarily in the middle of the planes for the USB traces, so there's still non-trivial metal around them.

I think it's about ready to pull the trigger on Rev B

closed 2016-09-29 by ppannuto