lab11/signpost #2: Controller RevA Updates

ppannuto, 2016-09-29:
Issue to track fixes for controller RevA

  - [ ] There are two pins labeled `USB_D+` and none `USB_D-`

brghena, 2016-09-29:
Other issues with Controller Rev A

  - [ ] VR1 doesn't have a Pin 1 marker
  - [ ] 
  - [ ] 
  - [ ] 
  - [ ]

ppannuto, 2016-09-30:
@brghena when you fix "D1 has basically no markings on it's package and is really hard to tell orientation on", can you poke me to fix on seinduino somehow?

brghena, 2016-10-01:
- [ ] GND pads extremely hard to solder on controller module headers

brghena, 2016-10-23:
- [ ] Need ability to reset GPS

brghena, 2016-11-02:
- [ ] USB_ID connection for Edison is incorrect. Needs to be Grounded in normal case (acts as host) and Floating in programming case (acts as peripheral). (https://www.maximintegrated.com/en/app-notes/index.mvp/id/1822)

nealjack, 2016-12-06:
- [ ] Edison on controller draws 2x-4x the current of an Edison on the breakout board
- [ ] Controller draws over 100mA on its own. We think it's the GPS
- [ ] Power gate GPS, add reset line
- [ ] Power gate everything

brghena, 2017-01-18:
The vast majority of this has been handled in Control Module rev B (ca164b0e88493279cb7291171baee575391ca474)

The issues with D1 and C25-C27 have been ignored. The large power draw of the Edison will be further investigated when the new board is complete

closed 2017-01-18 by brghena