lab11/signpost #4: Still having problems with Baud rate not matching the setting

adkinsjd, 2016-10-03:
In tock, I am still having to manually adjust my baud rate up to 125000 to get an actual baud rate of 115200.

bradjc, 2016-10-03:
Yeah I don't know what's up with that. Is it possible Michael did actually calibrate the RC32K clock for every firestorm, and we never actually overwrote the bootloader so it wasn't an issue? Do we have any untouched firestorms we could look at address 0xfe04 of?

prabaldutta, 2016-10-03:
Why are you using the RC oscillator for a BRG???  373 fail!  RC oscillators
have terrible tempco!

- Prabal

alevy, 2016-10-03:
On what I _think_ is an untouched Firestorm I'm reading `0x1d001a` from address `0xfe04`

alevy, 2016-10-03:
@prabaldutta The UART's baud rate generator is _not_ sourced from the RC32K, it's sourced from the CPU clock (which is the setup to use the DFLL).

@adkinsjd & @bradjc I'm not sure why we decided this might be related to the RC32K... is it possible the divider for the DFLL is off by a bit or something? Resulting in a frequency of about 44Mhz instead of 48Mhz?

bradjc, 2016-10-03:
`0x1d001a` differs from what we had (which was `0x1d0015`). It may be time to dig into the clock land again...

closed 2016-10-14 by bradjc