lab11/signpost #16: Backplane RevB updates

ppannuto, 2016-10-16:
So far just nits but a place to collect things.

 - [ ] Add a space on silkscreen to write in an identifier (i.e. Backplane no. 12)
 - [ ] Move `R46` and `R47` to be near `R48`. (Or vice-versa and move the test points too)

adkinsjd, 2016-10-17:
I2C bypass jumpers in case the isolators aren't working

adkinsjd, 2016-10-20:
- [ ] Large annular rings around all mechanical holes!
- [ ] Don't route any traces near mechanical holes

bradjc, 2016-10-26:
[ ] All traces on middle layers?

adkinsjd, 2016-11-17:
A generally thicker, more sturdy pcb

bradjc, 2016-11-23:
If the PCB gets thicker, we have to keep in mind the through-hole parts and make sure they are still solderable.

ppannuto, 2017-03-06:
- [ ] replace GPIO extender package with version from debug backplane

brghena, 2017-03-06:
- [ ] Add GPIO extender to reset / monitor the USB hub

ppannuto, 2017-03-07:
- [ ] Validate all component heights are < 0.080" (for compat w/ thicker pcb)

ppannuto, 2017-03-07:
> All traces on middle layers?

Where are we at on this - superfluous or worth the effort? I kinda don't want to do it just because it'll take a while to do, but if it's worth it I will

brghena, 2017-03-07:
I think it's unnecessary

adkinsjd, 2017-03-07:
I think as long as you are especially careful to keep traces away from
screws and edges you are fine. The removal of internal module seals should
make this even less of an issue

ppannuto, 2017-03-08:
I think I've now addressed everything here.

Do we have other changes or additions to the backplane?

ppannuto, 2017-03-10:
- [ ] Connect the watchdog from controller to power module
   - [ ] @brghena update controller module in library to add watchdog pin
   - [ ] @nealjack update power module in library to add watchdog pin

nealjack, 2017-03-17:
- [ ] Step up the vbat+/- and vsol+- traces to at least 100mil

closed 2017-06-26 by ppannuto