lab11/signpost #20: Summon demo app

brghena, 2016-10-23:
Thomas added a summon app which can be viewed at [j2x.us/signp](http://j2x.us/signp). This issue is for tracking updates that are needed for it. @tzachari 

- [ ] Add time and date to the `Controller` view
- [ ] Add the UC Berkeley logo to the signpost
- [ ] Either add the UCSD logo to the `Air Quality` view or rename to `UCSD Air Quality`
- [ ] Make `SIGNPOST` name bigger. Might have to move it

ppannuto, 2016-10-23:
- [ ] Resolve mixed security content

The j2x url currently redirects to https rawgit, but the websocket uri isn't over secured websockets. Manually going to the http version of the rawgit page allows the websocket to connect, but this should be consistent.

[image]

brghena, 2016-10-23:
- [ ] the '2' in 'CO2' doesn't load right on android
[image]

bradjc, 2016-10-24:
- [ ] add controller energy to power supply box

closed 2016-11-22 by brghena