lab11/signpost #22: Controller Rev B Fixes

brghena, 2017-01-26:
List for tracking problems discovered with Rev B of the controller

 - [ ] Swap C12 and C25 locations

brghena, 2017-01-27:
- [ ] Clearance on Edison mounting holes is insufficient
- [ ] Pull-up on LINUX_ENABLE_POWER is not working

ppannuto, 2017-02-14:
- [ ] Connect !BACKPLANE_RESET to a SAM4L

ppannuto, 2017-02-19:
- [ ] Label the usb ports on the *top* of the board

brghena, 2017-02-20:
- [ ] Better isolation for things touching the Linux power domain
- [ ] Mod_in/out pins for the Storage Master too

ppannuto, 2017-02-22:
- [ ] connect module i2c (and may as well do smbus while we're at it) pins to gpio inputs as well so controller can monitor

ppannuto, 2017-02-22:
- [ ] PPS should also be input to controller (and storage master while we're at it)

brghena, 2017-03-07:
- [ ] Interrupt pin to the Edison

brghena, 2017-03-07:
- [ ] Interrupt pin from Storage Master to Controller

adkinsjd, 2017-03-23:
Individual reset buttons for each MCU

adkinsjd, 2017-04-25:
- [ ] Add POR chips for both SAM4Ls. Backplane power nets are too large to fulfill necessary voltage rise rates, meaning SAM4Ls need POR.

closed 2017-06-28 by brghena