lab11/signpost #23: Debug Backplane Fixes

adkinsjd, 2017-01-31:
[ ] Switch module 1 and 0 programming silkscreen (light is still correct)

adkinsjd, 2017-01-31:
- [ ] Probably should put default off resistors on the debug LEDs

adkinsjd, 2017-01-31:
- [ ] Nit - put the programming signals in order
 - [ ] Move silkscreen out to accommodate a knurled knob

adkinsjd, 2017-01-31:
- [ ] A big 5V warning. maybe  a protection circuit

ppannuto, 2017-02-14:
Moving from #26 here:

 - [ ] Add LED for fake radio module.

ppannuto, 2017-02-14:
- [ ] Deal with USB Hub Reset problem

brghena, 2017-02-14:
## USB Hub Reset

Reboot: Hub still good
Power cycle (with ten seconds physically powered off): Hub still good

I'm guessing just asserting BackplaneReset at boot will be sufficient

bradjc, 2017-02-20:
[ ] Doesn't say "Signpost"
[ ] Module 0, Module 1, Controller font are not the same size.
[ ] "MEM" -> "Storage"
[ ] What is the empty programming switch position for?
[ ] Move the JTAG header to the top of the board.

brghena, 2017-02-20:
- [ ] No easily accessible ground post for probing

adkinsjd, 2017-02-20:
Literally every screw? Or the 4 gnd testpoints at various spots on the
board?

brghena, 2017-02-20:
Sorry, first message was unclear. I edited it to include the real intent, which is the ground connections on oscilloscope leads, which are too big and would short to other pins on the 0.1" headers and cannot latch on to round-headed screws.

ppannuto, 2017-02-20:
I'm a fan of the ground bar-like thing. Draw a rectangular poly of exposed ground metal along the edge so that leads can just clip to the pcb.

adkinsjd, 2017-02-20:
You should unscrew one of the feet for now

bradjc, 2017-02-21:
- [ ] Mod0 and Mod1 debug headers should be identical.

brghena, 2017-02-21:
- [ ] Enable programming of modules even without a valid controller app

Currently, when you just plug a module in to the debug backplane, you can't program it unless the controller has disabled isolation for that module. One solution to this would be powering modules explicitly even when the programming switch is set to it.

bradjc, 2017-02-21:
- [ ] Bigger !RESET button switches.

adkinsjd, 2017-02-21:
I am actually pretty strongly against this unless you can enable and
disable it strictly when the jlink is actually flashing. I think that a
module staying on when the controller wants it off would be more
problematic in the long run, and I don't think requiring turning the switch
to be selecting none of the boards is viable, as you would inevitably
forget.

There is probably a jlink signal that could switch on/off for the duration
of programming that could make this happen though.

ppannuto, 2017-02-21:
Yeah, the whole flashing / dev story needs a little work. Currently we have
the annoying problem where flashing one quite often tends to leave other
boards in the "i2c lines stuck low until reset" state that the SAM4L likes
to get into.

The big hammer is obviously to put all boards into reset during a flash,
power on the target module forcefully, and then take things out of reset -
but that could be obnoxious in practice. I'd rather understand this buggy
i2c state

adkinsjd, 2017-02-21:
> AGIJUD7p0tJnIDTsNDVoqEf99e0N0VOIks5re0m4gaJpZM4LyRZM

The buggy I2C state is a separate issue, which I worry we will never be
able to fix because it's a chip bug (although the first thing I would try
is adding a few extra jlink resets at the end of the flashing script).

ppannuto, 2017-02-21:
Yeah.. adding jlink resets won't help. It's _other_ SAM4Ls that aren't
being flashed that tend to hold low it in my experience.

adkinsjd, 2017-02-21:
> issuecomment-281470960
> GYQ9iCI-nMBks5re07XgaJpZM4LyRZM

Really? That's not what I've found. Usually if I hit the reset button on
the one I'm flashing it works.

adkinsjd, 2017-02-21:
> AGIJULLDDh1GodUsrCFE-GYQ9iCI-nMBks5re07XgaJpZM4LyRZM

I just assumed something about pressing the button reset the chip
differently

ppannuto, 2017-02-21:
Often is too strong. I'd say.. 75% of the time the one I flashed, maybe 25%
of the time another chip on the network somewhere. But at this point I'm
just in the habit of resetting all of them, so my empirical evidence isn't
great. Maybe I can unlearn that habit.

bradjc, 2017-02-21:
My experience is basically 100% of the time it's the fault of the one I just flashed and that resetting the just flashed module fixes the issue.

adkinsjd, 2017-02-21:
Have we tried just running JLinkExe and resetting a few times to see if
that fixes it?

ppannuto, 2017-02-21:
- [ ] Master reset button that resets everything

ppannuto, 2017-02-22:
- [ ] Another USB hub upstream of the FTDIs so that you can just plug in one USB cable

nealjack, 2017-02-22:
[ ] - Radio Module clock unstable - gpios don't work on cold boot

ppannuto, 2017-03-07:
- [ ] Add GPIO extender to reset / monitor the USB hub. Set to address 3

ppannuto, 2017-04-10:
- [ ] power metering jumpers for modules/controller
 - [ ] eeprom chip