lab11/signpost #28: Edison toFlash.tar.gz firmware fails to reboot

cxbrooks, 2017-02-15:
The signpost/software/edison/toFlash.tar.gz  image seems to not work for me. 

I installed git lfs:

```
sudo port install git-lfs
git clone https://github.com/lab11/signpost.git
cd signpost/software/edison                                                                                    
git lfs track toFlash.tar.gz                                                                                    
git lfs pull toFlash.tar.gz  
```

The image size seems correct and the file seems OK:

```
bash-3.2$ ls -l toFlash.tar.gz                                                                                            
-rw-r--r--  1 cxh  staff  158338761 Feb 15 11:39 toFlash.tar.gz                                                           
bash-3.2$ tar -ztvf *.gz | head                                                                                           
drwxrwxrwx  0 root   root        0 Nov 22 01:15 toFlash/                                                                  
-rw-rw-rw-  0 root   root  4194304 Nov 22 01:09 toFlash/edison_ifwi-dbg-01-dfu.bin                                        
-rw-rw-rw-  0 root   root  4194304 Nov 22 01:09 toFlash/edison_ifwi-dbg-00-dfu.bin                                        
-rw-rw-rw-  0 root   root      742 Nov 22 01:09 toFlash/filter-dfu-out.js                                                 
-rw-rw-rw-  0 root   root  4194468 Nov 22 01:09 toFlash/edison_ifwi-dbg-01.bin                                            
-rw-rw-rw-  0 root   root  4194304 Nov 22 01:09 toFlash/edison_ifwi-dbg-03-dfu.bin                                        
drwxrwxrwx  0 root   root        0 Nov 22 01:09 toFlash/helper/                                                           
drwxrwxrwx  0 root   root        0 Nov 22 01:09 toFlash/helper/images/                                                    
-rw-rw-rw-  0 root   root   315857 Nov 22 01:09 toFlash/helper/images/Edison-arduino-blink-led.png                        
-rw-rw-rw-  0 root   root   298970 Nov 22 01:09 toFlash/helper/images/Edison-breakout-board.png                           
bash-3.2$ 
```
When I use the Intel Edison Setup Application to install this file, I see it validate the image, and then there is a message about dfs (not sure about that) and then I get a window that says:

> Uh-oh! Your Intel Edison did not update, This is likely because your Intel Edison did not reconnect properly, You'll need to download the newer firmware and manually update the Intel Edison Board. Detailed instructions are at https://software.intel.com/en-us/flashing-your-firmware-edison

I'm able to download the latest image from Intel and install that.

Should signpost/software/edison/toFlash.tar.gz work on an bare Intel Edison that is just connected to a [SparkFun Base Kit for Intel® Edison KIT-14105](https://www.sparkfun.com/products/14105) ?

adkinsjd, 2017-02-15:
I have never used the Edison Setup Application.

Usually to install our images, we unpack the tar ball, then run the
flashall.sh script from terminal.

This takes a few minutes, and doesn't work every time, but usually works
within 1-3 tries. If you have problems and are in the DOP center, Neal or I
can look at it with you later today.

cxbrooks, 2017-02-16:
Awesome, that worked.

Follow t[he manual flashing instructions](https://software.intel.com/en-us/flashing-firmware-on-your-intel-edison-board-mac-os-x)

Assuming brew is already installed

```
brew install dfu-util coreutils gnu-getopt
cd signpost/software/edison/
tar -zxf toFlash.tar.gz
cd toFlash
./flashall.sh
```

In the Finder, unmount the Edison partition

Be sure to be plugged in to the otg port, not the console port of the [SparkFun Base Kit for Intel® Edison KIT-14105](https://www.sparkfun.com/products/14105)

Below is a sample run:

```
bash-3.2$ ./flashall.sh
Using U-Boot target: edison-blankcdc
Now waiting for dfu device 8087:0a99
Please plug and reboot the board
Flashing IFWI
Copying data from PC to DFU device
Copying data from PC to DFU device
Flashing U-Boot
Download        [=========================] 100%       237568 bytes
Flashing U-Boot Environment
Copying data from PC to DFU device
Flashing U-Boot Environment Backup
Copying data from PC to DFU device
Rebooting to apply partition changes
Now waiting for dfu device 8087:0a99
Flashing boot partition (kernel)
Copying data from PC to DFU device
Flashing rootfs, (it can take up to 5 minutes... Please be patient)
Copying data from PC to DFU device
Rebooting
U-boot & Kernel System Flash Success...
Your board needs to reboot to complete the flashing procedure, please do not unplug it for 2 minutes.
bash-3.2$
```
My notes are at https://www.icyphy.org/accessors/wiki/Notes/Edison

I'm closing this because I was able to flash the signpost image.

ppannuto, 2017-02-16:
I'm going to reopen this as a reminder to come back and improve the documentation at some point

commit da9ac3e0 (2017-02-21): Update README.md

 - Flesh out install more, esp faqs and issues
 - Clean up small language and other errors

Closes #28.
closed 2017-02-21 by ppannuto