lab11/signpost #30: Updates for the Edison image

cxbrooks, 2017-02-17:
Below are some updates I'd like to see in the Edison Image:

- [ ] Install Node and npm.  Node 6.9.5 is preferred.  See [Edison Node Notes](https://www.icyphy.org/accessors/wiki/Notes/EdisonNodeNotes).  Accessors will require Node.

- [ ] Set the date.  When I installed the default Intel-provided image, it seemed like the date was set properly, or at least close to the current time.  I ran into problems using wget and npm because the date was set to 1/1/2000 and the certs were not valid.

- [ ] Set up a sbuser with the appropriate keys, see [Set up the sbuser account on the Edison](https://www.icyphy.org/accessors/wiki/Notes/Edison#SetUpSbuser).  Downloading accessors requires the sbuser account.

There is no big rush on these, just when we get around to updating the Edison image, it would be good to consider these.

nealjack, 2017-02-17:
Great. This list is very useful. I'm actually working on making a more
complete image right now. If you think of any other packages you want
installed by default, please add them.

ppannuto, 2017-02-19:
These, in order:

 - [ ] apt install python3 python3-pip
 - [ ] apt install libffi-dev libssl-dev 
 - [ ] pip3 install --upgrade pip
 - [ ] pip3 install --upgrade setuptools
 - [ ] pip3 install python-periphery cryptography

ppannuto, 2017-02-19:
- [ ] apt install nfs-common

Related: mounting my machine's dev folder over nfs while working on the python edison stuff has made for a really nice working setup. This is just the client side and won't mount anything by default, so should be a perfectly sane/reasonable thing to include

ppannuto, 2017-02-19:
- [ ] give the edison user access to the I2C device

I recommend:
```
sudo chgrp netdev /dev/i2c-*
sudo chmod g+rw /dev/i2c-*
```

cxbrooks, 2017-02-19:
- [ ] (Optional) Write documentation about how the image was created.
- [ ] (Very optional) Create script that creates the image.

nealjack, 2017-02-20:
@cxbrooks in README.md there is a [link](https://jakehewitt.github.io/custom-edison-image/) to the process for building a debian image for the Intel Edison.

nealjack, 2017-02-20:
- [ ] udev rule for user i2c

ppannuto, 2017-02-20:
- [ ] set up https://github.com/lab11/signpost/blob/master/software/edison/support/edison-i2c.sh to run on every boot

ppannuto, 2017-02-22:
- [ ] apt install links

This is a text based browser that will give people a fighting chance of being able to navigate a captive portal.

ppannuto, 2017-03-01:
The only unchecked boxes are erpc stuff, but pretty sure @nealjack said that made it in - while it may need some love at some point in the future, that feels like a separate issue. The edison image as it currently stands is in a pretty good place.

closed 2017-03-01 by ppannuto