lab11/signpost #42: Generate mbedtls randomness from hardware RNG

alevy, 2017-02-25:
I believe the current implementation relies solely on CPU timing for randomness (specifically for the ECDH key exchange). That's a problem because it's insecure _and_ because we don't get to use our fancy TRNG driver for the SAM4L.

Instead, the signpost user libs can simply define 

```c
#define MBEDTLS_ENTROPY_HARDWARE_ALT

int mbedtls_hardware_poll( void *data,
                           unsigned char *output, size_t len, size_t *olen );
```

to call the underlying TRNG driver.

nealjack, 2017-02-25:
I've already written an [entropy abstraction](https://github.com/lab11/signpost/blob/master/software/apps/libs/signpost_entropy.c#L13) that uses the TRGN to as a source to seed the mbedtls PRNG.

alevy, 2017-02-26:
@nealjack Oh yeah, I think that works. I was looking through mbedtls, and there appears to be a supported way to do basically the same with a `#define` and magically named function you're meant to provide, but this seems better.

closed 2017-02-26 by alevy