lab11/signpost #43: Documentation: Map of repository

brghena, 2017-03-03:
It's unclear to new users what's located in which directory, and especially where the Tock submodule is located. Currently we point to Tock's getting started instructions, which assume you're in the Tock submodule to run commands. We should write a map of the repository and do something to make getting started make more sense to new users.

brghena, 2017-03-03:
We also need to include python3 installation instructions to support Tockloader and I guess python2 to support signpost-debug-radio

ppannuto, 2017-03-04:
I fixed signpost debug radio to support py2k or py3k. Only need 2k if you
care about the gdp

nealjack, 2017-03-16:
Also, update the module design specifications to indicate the dimensions and location of the debug header