lab11/signpost #44: Are you considering to use newer kernel on Edison?

andy-shev, 2017-03-04:
Just wondering if you are considering to switch to vanilla kernel at some point.

bradjc, 2017-03-06:
I'm not sure what you mean by "vanilla", but we would _love_ to use a newer kernel. If we could get to 4.5+ we could natively support the CC2520 that we have on some of our boards.

andy-shev, 2017-03-06:
[Vanilla](https://en.wikipedia.org/wiki/Linux_kernel#Development_model) is a standard name of upstream available kernel.

v4.11-rc1 supports almost everything important on Intel Edison out-of-the box. The progress on it is described on [community wiki](https://edison.internet-share.com/wiki/Using_a_vanilla_Linux_kernel_with_Intel_Edison#What_is_working_and_what_doesn.27t).

ppannuto, 2017-03-06:
@nealjack thoughts? It'd be great to jump up, but the missing power unit and power button support (esp the latter) give me some pause.

andy-shev, 2017-03-06:
@ppannuto 
To clarify a bit. There are two power related MCUs inside Atom SoCs:

1. PMU (in some it's called PMC), which takes care of South Complex devices (the driver _is_ in upstream!).
2. Power Unit is another one which controls North Complex devices, that are not used on Intel Edison anyhow (AFAIR), there is a debug support in upstream which allows you to see (read) the current state of those devices. That's why the support of it is low priority.

Regarding power button, I would ask Ingo Molnar or Thomas Gleixner to apply rest of the patches which are needed to enable it (actually one patch is needed) to next v4.11-rcX. I can't promise it will go to it, so v4.12 is for sure.

Don't hesitate to ask any question either by email (public mailing lists, Intel community forums, etc.) or thru GitHub.

nealjack, 2017-03-06:
When I find some time this week, I'll give the new kernel a try. I wasn't aware Intel was still maintaining the Edison kernel. I built the current Signpost image from what is available from [here]( https://downloadcenter.intel.com/download/25028/Intel-Edison-Board-Software-Package?v=t), and used a Debian rootfs.

andy-shev, 2017-03-06:
@nealjack 
Even that is old one. Latest official BSP v3.5 is dated 6.06.2016. Yeah, it has v3.10.98 based kernel, but at least a bit newer than initial one.

My kernel is unofficial. I'm doing this most likely as a hobby because I like the hardware and platform.
Same regarding U-Boot support and upstreaming.

andy-shev, 2017-03-20:
Power Button support is in v4.11-rc3 (and so in my branch here on GitHub).

vsobolyev, 2017-03-28:
Hi @andy-shev, 
I'd really wanted to try and use your kernel because of the SPI driver and DMA support.
Is there a way of building your kernel using the BSP v3.5 is dated 6.06.2016? I want this mostly because it seems easier to me to include new modules and try them through patches and bitbake.

andy-shev, 2017-03-28:
@vsobolyev 
It would be really appreciated, though I have 0 experience with Yocto (bitbake), so I dunno how to answer to your question.

vsobolyev, 2017-03-29:
@andy-shev 
Things were breaking as I was writing you hehe.
Thanx for the info. Will give try to the usual build later on.

andy-shev, 2017-03-29:
@vsobolyev 
You are welcome. One remark, it might be better to use issue tracker in my repository to avoid noise here. Since there is no automated way to move some comments as new issue in other repository, feel free to create one an put there comments that matters.