lab11/signpost #48: Module Time Sync

adkinsjd, 2017-03-18:
To have a hope of tight time synchronization, modules are probably going to need to wire PPS to a timer/counter (TC) pin, so that it can trigger a counter start. This might not really help, because the SAM4L doesn't have a 32bit counter that can take an external trigger anyways, but it's something to think about.

prabaldutta, 2017-03-20:
Why couldn't you connect it to a capture line on a free running timer?

- Prabal

adkinsjd, 2017-03-20:
The only 32bit free running timer peripheral does not have any way to
interface with external i/o in hardware, and its fastest clock source is
32khz. I was going to test full accuracy of time-stamping events in
software using this timer today.

I expect to achieve synchronization in the 10s of uS using this method,
which is probably good enough for something like localizing an audio
source, but still much worse than our original expectation of
synchronization given the accuracy of the pps line.

> To have a hope of tight time synchronization, modules are probably going
> to need to wire PPS to a timer/counter (TC) pin, so that it can trigger a
> counter start. This might not really help, because the SAM4L doesn't have
a
> 32bit counter that can take an external trigger anyways, but it's
something
> to think about.
lLeQVFRWu6WIw1NwqF5dGFwks5rm2Z7gaJpZM4MhTAW>

—
<https://github.com/lab11/signpost/issues/48#issuecomment-287780626>, or mute
the thread

prabaldutta, 2017-03-20:
Wow...that's interesting and a bit disappointing that the MCU is missing
this pretty basic functionality.  I'm surprised that there isn't a 32-bit
timer block that can be configured to be free running and that also gives
you a capture line.

- Prabal

adkinsjd, 2017-03-20:
Yeah, I was pretty shocked. Usually you can at least chain two 16bit timers
together in hardware, and I can't even find a way to do that.

I might resort to running our synchronization evaluation code on the
cc2538, but I haven't looked into it much yet.  Hopefully that would at
least allow us to demonstrate the capabilities of the platform, even if
most of the modules we currently have can't achieve that accuracy.

prabaldutta, 2017-03-20:
Probably not a bad idea.  Our module selection kinds undercuts the basic
claim.

It might be useful to just do this on some third-party module that runs its
own ecosystem (e.g. mbed, Arduino, TI, etc.) to demonstrate both the timing
and the modular compatibility.

- Prabal