lab11/signpost #49: Power Module Rev C issues

nealjack, 2017-03-22:
- [ ] U19 pin 1 indicator
- [ ] Better precision resistors for solar charger divider (Input and output regulation)
- [ ] More space monitors and sense resistors
- [ ] More space around charger/regulators
- [ ] U27 wrong package
- [ ] D3 wrong package
- [ ] Add reset button on watchdog

brghena, 2017-03-22:
Before you spin Rev D, it would be good to check if the watchdog firing actually turns off and back on the power rails correctly. (Probably a deadbugged chip and some wires)

nealjack, 2017-03-22:
U27 is the watchdog - so I don't have a good way

nealjack, 2017-03-22:
- [ ] VSOL- must be connected to VBAT-. Watchdog only shuts off battery connection to ground, solar panel could still power device. Change charger GND connections to VSOL-/VBAT-.

brghena, 2017-04-07:
- [ ] Think about Load Switches on a per module basis

brghena, 2017-10-10:
- [ ] WHY IS THE RESET BUTTON SO HARD TO PRESS??