lab11/signpost #54: signpost-debug-radio issue

brghena, 2017-05-23:
When Theo was running through the Signpost tutorial, he found that the signpost-debug-radio pip package seems to have an issue and can't find `_version.py`. We should try to replicate this and fix it. Leaving this issue for now so that I remember it.

brghena, 2017-05-24:
Fixed with https://github.com/lab11/signpost/commit/8633d549c00d0ca75cb1869d9ac14b1dbdb60efc. The issue was not looking for the local `_version.py` file. Version 0.2.1 of `signpost-debug-radio` has been released.

closed 2017-05-24 by brghena

ppannuto, 2017-05-24:
> Fixed with 8633d54

Does that work in both py2k and py3k? I thought you couldn't do dot imports
in py2?

brghena, 2017-05-24:
It's been tested in python 2.7 and works just fine.