lab11/signpost #56: Add a license

Susurrus, 2017-05-23:
While this code is all public, I didn't see a license file or anything specified in the README. It'd be good to clarify this.

adkinsjd, 2017-05-23:
It's free, and you can use it if you want to. I suppose you could take that as the "Unlicense".

Susurrus, 2017-05-23:
Cool, that's a great license choice! Could this be made explicit in the `README` or in a `LICENSE` file so that it's clear for others as well?

closed 2017-05-24 by adkinsjd

Susurrus, 2017-05-24:
I believe this repo also contains hardware, which doesn't quite work with some of the conventional FOSS software licenses. I'd recommend explicitly selecting a specific OSHW license for your hardware designs and then specifying that only the code is under Apache2/MIT.

ppannuto, 2017-05-24:
My reading / understanding was the MIT (and/or BSD) licenses covered hardware well. The copyleft fellows had to change but not the permissive ones. A quick google seemed to back this understanding, but it seems to be murkier territory. If you have pointers to any good writeups on the issues with using MIT license for OSHW / recommended licenses we'd be happy to consider them.

Susurrus, 2017-05-24:
[This seems like a good resource](https://www.oshwa.org/sharing-best-practices/). But basically, MIT shouldn't be an issue, you're correct. You might want to say that non-code in this repo is licensed under MIT and all code is dual MIT/Apache2. Then I think you'd be fully covered.

PR #57 (2017-05-24): Formally license
This is all the legal boilerplate that's made other projects (and, importantly,
their sponsors) happy.

Closes #56.
commit 0b26a9ca (2017-05-24): Formally license

This is all the legal boilerplate that's made other projects (and, importantly,
their sponsors) happy.

Closes #56.